SciLor/Hyperion_LED-Controller #5: Compile Error

SquallHalle, 2016-11-13:
Arduino 1.6.12
ArduinoJason by Benoit Blanchon 5.7.0
ArduinoThread by Ivan Seidl 2.1.0
Linked List by Ivan Seidel 1.2.3
FastLED by Daniel Garcia 3.1.3
Logging von dir im libaries ordner. 

 -mkdir testordner
 -cd testordner
 -git clone https://github.com/SciLor/ESP8266_Hyperion_LED-Controller.git
 -Arudino IDE öffnen ~/testordner/ESP8266_Hyperion_LED-Controller/HyperionRBG.ino
 -Medlung: Die Datei "HyperionRBG.ino" muss sich in einem Sketch-Ordner befinden. Soll der Ordner angelegt, die Datei verschoben und dannach fortgefahren werden? -> OK
 -Sketch ist offen
 -cp ConfigStatic.h.example ConfigStatic.h 
 -editor ConfigStatic.h , anpassen
 - mv * ~/testordner/ESP8266_Hyperion_LED-Controller ~/testordner/ESP8266_Hyperion_LED-Controller/HyperionRGB (das ist der von der IDE angelegte SketchOrdner)
 - Arduino IDE -> überprüfen (Kompilieren) 

Ausgabe: 
	In file included from ~/testordner/ESP8266_Hyperion_LED-Controller/HyperionRGB/HyperionRGB.ino:14:0:
sketch/WrapperJsonServer.h:6:24: fatal error: WifiServer.h: No such file or directory
 #include <WifiServer.h>
                        ^
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

SciLor, 2016-11-13:
Dir scheint "WifiServer.h" zu fehlen. Dies sollte in der Lib für den ESP8266 drin sein. Also ist es ein Problem mit deiner Build Umgebung.
Hast du "http://arduino.esp8266.com/stable/package_esp8266com_index.json" in den Voreinstellung als Boardverwalter mit drin und esp8266 2.3.0 installiert?

SquallHalle, 2016-11-14:
Klar ist die LIB vorhanden, dass ist die ESP8266WiFi.

#include <WiFiClient.h>
#include <WiFiServer.h>
#include "BaseHeader.h"
#include <ArduinoJson.h>
//#include "WifiServer.h"
//#include "WifiClient.h"
wenn man deinen Aufruf der Headerfiles anpasst, von "schau in den Sketchordner" zu "schau in den LIB-ordner", dann findet er sie auch. 

Dennoch kommt es zu Fehlern[1], so dass ich annehme du benutzt eine andere LIB ?

[1]
```
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `ClientContext::getSize() const':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::write(unsigned char)'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:165: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::~WiFiClient()':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::peek()'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:234: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::peekBytes(unsigned char*, unsigned int)':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::connected()'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:278: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::operator bool()':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::operator bool()'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:294: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::connect(char const*, unsigned short)':
Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: /home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi
 Nicht benutzt: /home/user/IDE/ArduinoIDE/libraries/WiFi
 Nicht benutzt: /home/user/IDE/ArduinoIDE/libraries/WiFi
 Nicht benutzt: /home/user/IDE/ArduinoIDE/libraries/WiFi
 Nicht benutzt: /home/user/IDE/ArduinoIDE/libraries/WiFi
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::connect(char const*, unsigned short)'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:89: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o:(.bss._ZN10WiFiClient10_localPortE+0x0): multiple definition of `WiFiClient::_localPort'
sketch/WiFiClient.cpp.o:/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Stream.h:92: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::_s_err(void*, signed char)':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::_s_err(void*, signed char)'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:335: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::_s_connected(void*, void*, signed char)':
WiFiClient.cpp:(.text._ZN10WiFiClient12_s_connectedEPvS0_a+0x4): multiple definition of `WiFiClient::_s_connected(void*, void*, signed char)'
sketch/WiFiClient.cpp.o:WiFiClient.cpp:(.text._ZN10WiFiClient12_s_connectedEPvS0_a+0x4): first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::connect(IPAddress, unsigned short)':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::connect(IPAddress, unsigned short)'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:99: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::available()':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::available()'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:207: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::peekBytes(unsigned char*, unsigned int)':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::peekBytes(unsigned char*, unsigned int)'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:241: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::stop()':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::stop()'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:269: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o:(.bss._ZN5SListI10WiFiClientE8_s_firstE+0x0): multiple definition of `SList<WiFiClient>::_s_first'
sketch/WiFiClient.cpp.o:/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Stream.h:92: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::~WiFiClient()':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include/ClientContext.h:276: multiple definition of `WiFiClient::~WiFiClient()'
sketch/WiFiClient.cpp.o:sketch/WiFiClient.cpp:62: first defined here
libraries/ESP8266WiFi/WiFiClient.cpp.o: In function `WiFiClient::~WiFiClient()':
[... cut ...]
sketch/WiFiServer.cpp.o:sketch/WiFiServer.cpp:159: first defined here
libraries/ESP8266WiFi/WiFiServer.cpp.o: In function `WiFiServer::available(unsigned char*)':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiServer.cpp:99: multiple definition of `WiFiServer::available(unsigned char*)'
sketch/WiFiServer.cpp.o:sketch/WiFiServer.cpp:99: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).
```

SquallHalle, 2016-11-14:
Das kompiliert im übrigen sauber: 
```

#include <ESP8266WiFi.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

SciLor, 2016-11-14:
Probiere bitte einmal eine saubere neue Installation von der IDE. Es muss mit den spitzen Klammern funktionieren. 
Ich hatte die Readme dementsprechend mit den Versionsnummern aktualisiert.

SquallHalle, 2016-11-14:
habe es schon 2 mal sauber neuinstallier, alle configs und so entfernt, neu das zip geladen entpackt und so weiter

SciLor, 2016-11-14:
Wenn ich das anlegen mal nachvollziehe, die HyperionRGB.ino verschieben lassen einen Unterordner. Dieser Unterordner enthält aber nur die HyperionRGB.ino.
Hast du die restlichen Dateien mitgezogen?
Ich bekomme ansonsten
`test\HyperionRGB\HyperionRGB.ino:4:24: fatal error: BaseHeader.h: No such file or directory

 #include "BaseHeader.h"`

SquallHalle, 2016-11-14:
klar hab ich die dann nachverschoben, es liegen alle datein nun in dem ordner

SciLor, 2016-11-14:
Bist du auf Linux unterwegs?

SquallHalle, 2016-11-14:
jub. ubuntu 16.10

SciLor, 2016-11-14:
Mich macht stutzig, dass du in der einen Variante als Meldung bekommst "sketch/WiFiServer.cpp.o:sketch/WiFiServer.cpp:99: first defined here"
Ich probiere es auch mal auf Linux aus

SquallHalle, 2016-11-14:
...unter windows klappt alles wie gewünscht.

SciLor, 2016-11-14:
Auf Linux klappt es nicht, warum auch immer, ich mache noch eine Issue auf Englisch hier auf

SciLor, 2016-11-14:
#11

closed 2016-11-14 by SciLor

SquallHalle, 2016-11-14:
öhm, ich sag mal: freut mich. good bug hunting ! und danke !

SquallHalle, 2016-11-14:
kann man dich noch anderweitig erreichn, ich hätte da zur config mal noch ne frage

SciLor, 2016-11-14:
Schau mal hier rein: http://www.forum-raspberrypi.de/Thread-hyperion-tutorial-esp8266-nodemcu-addon-wifi-led-controller-udp

SquallHalle, 2016-11-14:
schon gesehen wollte mich da jetzt nicht extra anmelden.