SciLor/Hyperion_LED-Controller #14: WS2812b led strip flicker

shaq80, 2016-11-18:
WS2812b led strip flicker/flashes towards the end of the strip.

solved by adding:
#define FASTLED_ALLOW_INTERRUPTS 0 to the line immediately before #include <FastLED.h>
in WrapperFastLed.h¸

reference:
https://github.com/FastLED/FastLED/issues/306

SciLor, 2016-11-18:
Could you try to add this define in the ConfigStatic.h 
This would be the best place to add that.
I would place it in the example Config then

shaq80, 2016-11-18:
I will try that later this evening/tomorow and get back to you with result.

shaq80, 2016-11-18:
i have added in ConfigStatic.h
#define FASTLED_ALLOW_INTERRUPTS 0
bellow #define CONFIG_LED_CLOCKLESS_CHIPSET WS2812B

it is working normal without flickering

SciLor, 2016-11-19:
Added it in
#15 
Thank you!

closed 2016-11-19 by SciLor