SciLor/Hyperion_LED-Controller #18: RGB 5050 support

iLLiac4, 2016-11-21:
Hi. I am looking into your code and i will try to build it.
Is it possible to add a support for single rgb stripe like this one https://github.com/iLLiac4/Adalight_Hyperion/blob/master/Adalight_RGB.ino with config optimized for it https://github.com/iLLiac4/Adalight_Hyperion/blob/master/hyperion.config.json.RGB connection scheme is something like this one http://www.jerome-bernard.com/images/rgb-led-strips-mosfets.png but it is possible to just use premade rgb 1usd controller and wiretap rgb onto it.

It is simple rgb contoller for adalight and works with hyperion.

I saw on internet esp2866 wifi led controller like this one http://www.banggood.com/ARILUX-AL-LC01-Super-Mini-LED-WIFI-Smart-RGB-Controller-For-RGB-LED-Strip-Light-DC-9-12V-p-1058603.html?rmmds=search i think it would be interesting to mod it with your code and if possible add rgb stripe support. https://mobile.twitter.com/electromodule/status/769092596044009473

SciLor, 2016-11-21:
Sure why not. I am going to add another LED Type PWM. Thanks for the example code.

iLLiac4, 2016-11-21:
cool. There are pwm outputs on esp2866 wemos http://i1.wp.com/www.esp8266learning.com/wp-content/uploads/2016/09/wemos-and-rgb-led_bb.png?w=513 for reference.

SciLor, 2016-11-21:
Please check out the latest version in the dev branch. I am unable to test it. Let me know if it works

iLLiac4, 2016-11-21:
Ok will solder it over the weekend.

iLLiac4, 2016-11-21:
Can you please pastebin both config; hyperion.config.json and the second one from esp2866 hyperion.config2.json. TNX

SciLor, 2016-11-21:
Both configs are for hyperion and not for the esp. They are also very individual. 
The only thing you need to do if you want to forward the led data, set device to udp and protocol to 0. 
Please check out hyperion-project.org for help with hyperion.

iLLiac4, 2016-11-21:
I know that but is it possible to forward it just with one hyperion like this one: 
http://pastebin.com/EFnesTHq

So hyperion will output lights to esp2866 and this is the only ambilight, not as second one.

SciLor, 2016-11-21:
As I said, this is a question related to hyperion itself, not about the esp. You need to set following device parameters:
```
{
      "colorOrder" : "rgb",
      "maxpacket" : 1450,
      "name" : "AmbiSZ-ESP8266",
      "output" : "192.168.1.250:19446", ///
      "protocol" : 0,
      "rate" : 250000,
      "type" : "udp"
},
```

closed 2016-11-21 by SciLor

iLLiac4, 2016-11-21:
TNX. It is working.