SciLor/Hyperion_LED-Controller #23: Arilux_AL-LC0X

iLLiac4, 2017-01-08:
Is it possible to add support for this device and keep the original function of it?
The original function is:
https://github.com/mertenats/Arilux_AL-LC03
More info about hardware:
https://www.muzik.ca/2016/11/30/hacked-esp8266-rgb-led-controller/

SciLor, 2017-01-10:
What do you exactly need from that device?
Using the LEDs may be possible with the right configuration.

iLLiac4, 2017-01-11:
It is device with already integrated voltage regulator so it is easy to use. The original functionality is like any other rgb led controller but with wireless. I was thinking if it is possible to use both functions at the same time; keep the original function and also have an option for hyperion from your git.

SciLor, 2017-01-11:
Sorry I cannot easy copy one to another. If you miss any features that my code doesn't have I can implement it.

closed 2017-01-11 by SciLor