SciLor/Hyperion_LED-Controller #24: [Question] ESP32 compatible?

qwasy, 2017-12-29:
Hi
Is it possible to make the software compatible with ESP32? They are more powerful and cost mostly the same^^
I tried compiling it myself, but it got an error in EnhancedThread...

SciLor, 2017-12-29:
It may work. If you have an error in Enhaced Thread you may missing some libaries. without full error logs I cannot help

qwasy, 2017-12-30:
So here's the error log, one normal and one with more infos.
[esp32_log.txt](https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/files/1594439/esp32_log.txt)
[esp32_extended_log.txt](https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/files/1594440/esp32_extended_log.txt)

What I mostly did, was changing the ESP8266 parts to ESP32, like the Wifi and such. And for the ESP8266 does it compile correctly.

SciLor, 2017-12-30:
Looks like your a missing a lib.

Thread cannot be found

mabushey, 2017-12-30:
Just want to comment I would like to do the same, this thread is very informative, thank you. :)

qwasy, 2017-12-30:
weird...
I have the same ArduinoThread installed as in your requirements...

SciLor, 2017-12-31:
Please check that you have got the Thread class available in your build enviroment. It looks like its missing.

qwasy, 2018-01-01:
Oh wow. 
It does compile now^^
I changed all #include <Thread.h> to #include "Thread.h"
and put the files in the folder, and now it works. Dunno why.. the Thread.h is the same as in the library...
Weird...
Okay, so I'll just need to wait for my ESP32 to be delivered^^

rudelm, 2018-01-03:
@qwasy can you create a PR or point to your Repo for the ESP32 version? would be really cool if this lib supports both types.

qwasy, 2018-01-05:
@rudelm yeah no problem^^
https://github.com/qwasy/ESP32_Hyperion
I hope that's okay for @SciLor

SciLor, 2018-01-05:
What about doing a clone of my rep and sending me a push about your changes? This way I can integrate it.

qwasy, 2018-01-05:
never done that before^^' but I'll try

SciLor, 2018-01-05:
You need to clone my repository, clone it to your desktop and overwrite all files with yours and commit it and push it back to your own repo. Then you can Push them over to my repo

qwasy, 2018-01-05:
okay, I think I got it working now^^
thanks^^

seuffert, 2018-01-09:
nice - with the changes introduced by qwasy i can confirm it is working on an esp32 :) (100 LEDs WS2801) and controlling it over UDP

qwasy, 2018-01-09:
Awesome^^
My ESPs did arrive as well, and it works with APA102^^
But, did the Webinterface work for you?
Mine did not...

I got this Message all the time:

`
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
I:  Start logging
I:  Configuration read at 0x0 with v3
I:  CFG=EEPROM config loaded
I:  Reset Mode
I:  Connecting to WiFi 
I:  Using dynamic ip
E: WiFi Connection Failed! Rebooting...
`

Maybe the ESP32 Webserver Library I use has a problem...
I mean the config is easy configurable by hand, the webinterface is just a bonus^^

SciLor, 2018-01-09:
great. Thank you for push. I will integrate it when I have found a way to use the thread library without coping it.

SciLor, 2018-03-10:
I have ordered 2 Wemos LOLIN32 to test with.

SciLor, 2018-05-05:
Compatible with esp32 with v0.9.0
See https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/releases/tag/0.9.0

closed 2018-05-05 by SciLor