SciLor/Hyperion_LED-Controller #28: request

drc85, 2018-01-06:
hi, im not sure if this is the right place but i have a request.

is it possible to implement this webserver inside of your project? or vice versa?
https://github.com/jasoncoon/esp8266-fastled-webserver

im  having two diffrent led in my appartment both are controlled from a wemos d1 mini...
i gave both static ips and added a button link in the webserver... 
so i controll  all of them with ONE SINGLE webserver.

im planning on on doing the same for the ambilight. i want the ambilight controlled over the wemos too..so the ambilightis also is usable as a normal lamp even if the HYPERION service isnt running (or my udoox86 is sleeping) this should be possible if there is way to get the udp signal inside of a pattern coded to show the ambilight..

i know how to create a new pattern called ambilight but it is only empty it has no function i dont know how to code just how to copy paste...  if there is a way i could get the ambilight pattern work as a udp receiver shouldnt it work then? 

what are the main files for getting the esp act as a udp receiver?
the wrapperjsonserver looks like it is the main file for the udp server...but im not sure :D
there is alot of stuff like logging and esp config wich is cool but its hard for a beginner to see wich one are neccesarry to get just the udp signal from hyperion showing up. so i can implement it into the webserver from the link..

thx

SciLor, 2018-01-07:
I don't think it is the best way to implement that in the Hyperion LED Controller.

I think you should write it from scratch up. This one here is to be controlled over the Hyperion App.

closed 2018-01-07 by SciLor

drc85, 2018-01-07:
i want to writte my own thats why i want to know what parts in the sketch are necessary for the upd transfer.

SciLor, 2018-01-08:
The implementation here is just for listening for a very specific UDP stream for the LEDs
https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/blob/master/HyperionRGB/WrapperUdpLed.cpp