SciLor/Hyperion_LED-Controller #29: Add relay

begicj, 2018-01-21:
Can You add a relay that is activated when the LED strip lights up, and goes off 1 minute after the LED strip is off?

SciLor, 2018-01-21:
What do you exactly mean?

closed 2018-01-27 by SciLor

begicj, 2018-03-07:
I wonder if it is possible to add the possibility that the RELAY that is connected to the ESP8266 automatically goes ON when the led strip lights up, lets say if the STATIC mode or UDP mode is ON and there is some colors on LED strip?

If UDP mode or STATIC mode activates the black color (or FastLED.clear command), and there is no light on LED strip for 1 minute, that the relay automatically goes OFF and remains OFF until the udp or static mode activates any color of the LED strip?

SciLor, 2018-03-10:
I don't understand you. What do you mean with "relay"?

begicj, 2018-03-12:
[image]

It will be used to switch ON ambient lamp (220 V) behind armchair when ESP8266 lights LED strip....

SciLor, 2018-03-15:
Sorry, that's not fitting in the project. You could easily implement it yourself, by adding code for switching your relay here:

[HyperionRGB.ino](https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/blob/master/HyperionRGB/HyperionRGB.ino)

check out the chamgeMode function.