SciLor/Hyperion_LED-Controller #30: [Question]leds light up white but no change color from hyperion

ellisium, 2018-01-30:
Hello,

1/led strip is ok (tested on arduino mega with fastled)
2/ wifi connection ok & hyperion_udp connection ok => log_verbose show RGB data (nodemcu is only connected on usb to arduino IDE serial monitor 
3/ led strip + nodemcu => leds turn on and light up white

issue: Leds doesnt change color based on hyperion remote or kodi. Stick on white

wired nodemcu D7 
board : nodemcu v3 lolin
leds: WS2812B
arduino IDE : 1.8.4
ES8266 : 2.3.0
hyperionRGB: 0.8.4

Thx for helping

/*------------------------------------------------*/
/*Configuration type*/

#define CONFIG_ENABLE_WEBCONFIG 1

//Replaces all values in the webconfig with the ConfigStatic.h values when CONFIG_ENABLE_WEBCONFIG is enabled
//#define CONFIG_OVERWRITE_WEBCONFIG 1

/*------------------------------------------------*/
/*Logging level*/

//#define LOGLEVEL LOG_LEVEL_INFOS
//#define LOGLEVEL LOG_LEVEL_DEBUG
#define LOGLEVEL LOG_LEVEL_VERBOSE

/*------------------------------------------------*/
/*Main static configuration*/
//This cannot be changed via web inteface

//#define CONFIG_LED_SPI_CHIPSET WS2801 //Comment out for clockless
#define CONFIG_LED_CLOCKLESS_CHIPSET WS2812B //Comment in for clockless
//#define FASTLED_ALLOW_INTERRUPTS 0 //Comment in if clockless stripe (ex. WS2812B) is flickering
//#define CONFIG_LED_PWM 1 //Comment in if PWM Stripe

#define CONFIG_LED_DATAPIN D7 //Comment out for PWM
//#define CONFIG_LED_CLOCKPIN D2 //Comment out for clockless / PWM
//#define CONFIG_LED_PWM_RED D1 //Comment in for PWM
//#define CONFIG_LED_PWM_GREEN D2 //Comment in for PWM
//#define CONFIG_LED_PWM_BLUE D3 //Comment in for PWM

//Pin order, see FastLED doc. NodeMCU should work with FASTLED_ESP8266_RAW_PIN_ORDER
#define FASTLED_ESP8266_RAW_PIN_ORDER
//#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
//#define FASTLED_ESP8266_D1_PIN_ORDER

#define CONFIG_LED_COLOR_ORDER RGB
#define CONFIG_LED_COUNT 63

//OFF, HYPERION_UDP, STATIC_COLOR, RAINBOW, FIRE2012
#define CONFIG_LED_STANDARD_MODE STATIC_COLOR
#define CONFIG_LED_HYPERION_AUTOSWITCH true 
#define CONFIG_LED_STANDARD_MODE_TIMEOUT_MS 5000 

/*------------------------------------------------*/
/*Main configuration*/
//You can leave it empty and override it via the web interface on port 80
#define CONFIG_WIFI_SSID "XXXX"
#define CONFIG_WIFI_PASSWORD "XXXX"

#define CONFIG_WIFI_STATIC_IP 1
const byte CONFIG_WIFI_IP[] = {192, 168, 1, 200};
const byte CONFIG_WIFI_SUBNET[] = {255, 255, 255, 0};
const byte CONFIG_WIFI_DNS[] = {192, 168, 1, 1};

#define CONFIG_WIFI_HOSTNAME "ESP8266"

#define CONFIG_PORT_JSON_SERVER 19444
#define CONFIG_PORT_UDP_LED 19446

SciLor, 2018-02-01:
You may use the wrong pins or the node is defect. Please try it with FastLED alone.

ellisium, 2018-02-04:
Hi, thx for reply.

Yes I did the test with arduino mega and fastled. test is successful alternate light red /green / blue
Yes I did the test with 3 different nodemcu lolin v3 (all new) and fastled (same code than arduino mega) test is failed, leds light up white

maybe PIN def issue?

Any help welcome.
Thx very much

SciLor, 2018-02-07:
So this is not a problem with my code.

Please try to get it running with a vanilla example of FastLed.

closed 2018-02-07 by SciLor

ellisium, 2018-02-14:
I got it working with nodemcu and ws2812b and basic fastled sketch powering led strip directly from the nodemcu using 3.3v and gnd and pin 2
same config with your code show led stuck on white. I tried all pin orders possible... not working

I tried also basic sketch fastled using external power supply  and led stuck white too

Any idea?

thx very much for your help

SciLor, 2018-02-14:
You should not power your LED-Stripes over the nodemcu 3.3V/5V. That may be one problem.

The other problem could be, that you are missing a common ground, so all GNDs are connected (LEDs, Powersupplies, NodeMcu)

ellisium, 2018-02-14:
it's wired like this : https://i.stack.imgur.com/bdonO.png

ellisium, 2018-02-14:
Ok resistor on data pin was the root cause.

maybe a schema wiring in readme should be welcome.

Issue solved.

Big thanks for your help and your great work