SciLor/Hyperion_LED-Controller #31: Use of deleted function

vKnmnn, 2018-03-16:
~/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.h:69:7: note: 'ESP8266WebServer::ESP8266WebServer(const ESP8266WebServer&)' is implicitly deleted because the default definition would be ill-formed: class ESP8266WebServer

You probably already are aware of this.

changing the line ```ESP8266WebServer _server = ESP8266WebServer(80);``` to ```ESP8266WebServer _server;``` and disabling webconfig at least does compile.

SciLor, 2018-03-16:
The problem is known and described in the Readme

Please use 2.3.0 of the arduino library, until it is fixed. 
a) esp8266 2.3.0 http://arduino.esp8266.com/stable/package_esp8266com_index.json - add to board urls - Version > 2.3.0 don't work

SciLor, 2018-05-05:
Fixed with https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/commit/309d129bc7474b4bea6c810e75b63e4da1c8edc8

closed 2018-05-05 by SciLor