SciLor/Hyperion_LED-Controller #33: Utilize Bluetooth of ESP32

SciLor, 2018-04-27:
Use the bluetooth module of the esp32 to control the lights.

Yous13, 2018-04-28:
Hi Scilor,

Thank you to take in consideration my request about the bluetooth. In fact recently i saw a member post on hyperion-project. He asked if output the Data over Bluetooth instead Wifi was possible and Redpanther told him that he may do it in theory https://hyperion-project.org/threads/data-output-via-bluetooth.402/

The only way was to use an Arduino and a HC-05 or HC-06 Module. Now with the ESP32 it will be easiest to make the same. ( Maybe not... ) 

As i told you Adalight Bluetooth is possible https://github.com/alanwoolley/adalight_bluetooth but as you can see the code is not for our ESP.

Note that i have these 2 modules at home but i haven't test yet.

So my idea is to use /dev/rfcomm0 as output, make an association with the ESP32 ( how ? command line, web page, apps ? maybe you have an idea on that too ) and transfer Data over Bluetooth instead Wifi. ( maybe use Wifi for the configuration / association ? )

I understand the code, sometime i can do a little change on it for my personal use but i'm not a developer lol. I don't know the better way, all the possibility but i am conviced that it's possible.

I saw many time users asking for that functionality so i think it will be fantastic if we can ( mostly you ) develop the code to make it working.

Thanks again and sorry to not helping a lot. What do think about it ? Possible or not ?

AmbiMod.

SciLor, 2018-04-28:
I think this should be fairly easy, but I see no use case, where WiFi isn't availabile. 
The Adalight Sketch has everything that is needed to implement it for the esp, except bluetooth. Bluetooth on the ESP32 is new to me, as I just got my first ESP32 (lolin).
If there is somebody really want to use it I consider to implement it.

Yous13, 2018-04-28:
Ok great. If it's easy to convert the code for an Esp it's a good beggining.

If you can take a look on this, maybe that can help you to understand how to use Bluetooth on ESP32 to control LedStrip. 
https://www.reddit.com/r/esp32/comments/77ior2/esp32_arduino_ble_neopixel_control/

Here another sketch for Adalight Bluetooth http://roboforum.ru/forum86/topic16303.html. In russian but the sketch is in english lol.

Too bad that i can't help you to write the code. I try to mix a sketch with another without luck, i've dream too much i guess.