SciLor/Hyperion_LED-Controller #40: Not an issue but need some help with config static mod

Yous13, 2018-05-10:
Hi scilor,

It's me again lol. Today i tried to mod your config static for my personal use but i have a little problem with variable.

Here you can see my code where i added some more things.

`#define WEBCONFIG
#define LOGLEVEL LOG_LEVEL_INFOS

#define CONFIG_WIFI_HOSTNAME "ESP8266" 
#define CONFIG_PORT_JSON_SERVER 19444
#define CONFIG_PORT_UDP_LED 19446
 
#define CONFIG_WIFI_SSID "ssid"
#define CONFIG_WIFI_PASSWORD "password"
#define OTAPASSWORD "AmbiMod"

#define CONFIG_WIFI_STATIC_IP 1
  #ifdef CONFIG_WIFI_STATIC_IP
    const byte CONFIG_WIFI_IP[] = {192, 168, 1, 123};
    const byte CONFIG_WIFI_SUBNET[] = {255, 255, 255, 0};
    const byte CONFIG_WIFI_DNS[] = {192, 168, 1, 1};
  #endif
  
#define TYPE_LED WS2812B
#define PWM_LED

#define CONFIG_LED_COUNT 24
#define CONFIG_LED_COLOR_ORDER RGB
#define CONFIG_LED_DATAPIN 7
#define CONFIG_LED_CLOCKPIN 5

#define FASTLED_ESP8266_RAW_PIN_ORDER
//#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
//#define FASTLED_ESP8266_D1_PIN_ORDER
 
//#define FASTLED_INTERRUPT_RETRY_COUNT 0
//#define FASTLED_ALLOW_INTERRUPTS 0 

//OFF, HYPERION_UDP, STATIC_COLOR, RAINBOW, FIRE2012
#define CONFIG_LED_STANDARD_MODE RAINBOW 
#define CONFIG_LED_HYPERION_AUTOSWITCH true 
#define CONFIG_LED_STANDARD_MODE_TIMEOUT_MS 5000`

#ifdef WEBCONFIG
  #define CONFIG_ENABLE_WEBCONFIG 1
  #define CONFIG_OVERWRITE_WEBCONFIG 1
#endif

I want to change LEDs type definition method by adding PWM_LED and SPI_LED variable.
I change your WrapperLEDControl by this :
`#ifdef PWM_LED
    FastLED.addLeds<TYPE_LED, CONFIG_LED_DATAPIN, CONFIG_LED_COLOR_ORDER>(leds, _ledCount);
  #else
    FastLED.addLeds<TYPE_LED, CONFIG_LED_DATAPIN, CONFIG_LED_CLOCKPIN, CONFIG_LED_COLOR_ORDER>(leds, _ledCount);
  #endif`

Compilation was working only if SPI_LED is set and when i indicate PWM_LED i have errors message.
Maybe more modification are needed on WrapperLedControl.h to match my Config.static.

If you can help me to resolve it.

Thanks. AmbiMod.

SciLor, 2018-05-11:
Could you post the error messages?
I currently not understand what's your intention.

Yous13, 2018-05-11:
Thank You Scilor for your answer but i figure out my mistake lol.

Here my final code for LED part

`#define PWM_LED                                            // Change with PWM_LED or SPI_LED.
#define CONFIG_LED_SPI_CHIPSET APA102
#define CONFIG_LED_CLOCKLESS_CHIPSET WS2812B

#define CONFIG_LED_COUNT 24
#define CONFIG_LED_COLOR_ORDER RGB 
#define CONFIG_LED_DATAPIN D7
#define CONFIG_LED_CLOCKPIN D4

// If PWM, to solve flickering problem.
//#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0 `

and what i changed on WrapperLEDControl.

` #ifdef PWM_LED
    FastLED.addLeds<CONFIG_LED_CLOCKLESS_CHIPSET, CONFIG_LED_DATAPIN, CONFIG_LED_COLOR_ORDER>(leds, _ledCount);
  #else
    FastLED.addLeds<CONFIG_LED_SPI_CHIPSET, CONFIG_LED_DATAPIN, CONFIG_LED_CLOCKPIN, CONFIG_LED_COLOR_ORDER>(leds, _ledCount);
  #endif`

In this configuration I only need to change PWM_LED to SPI_LED or Vice_Versa, the entire configuration keep the same. Don't need to comment anymore the Clockpin for PWM or Chipset...

I don't know if it's better than your configuration but i like my new settings lol. Here my entire config static if someone is interested.

`#define WEBCONFIG                                           // Activation de la page de configuration WEB. Commenter pour desactiver cette fonction
#define LOGLEVEL LOG_LEVEL_INFOS                            // Activation du LOG INFOS. Remplacer LOG_LEVEL_INFOS par LOG_LEVEL_DEBUG ou LOG_LEVEL_VERBOSE

                                                                                                        // --- Configuration Wifi et OTA --- //                                                                                                        
#define CONFIG_WIFI_HOSTNAME "ESP8266"                      // Nom du Dispositif
#define CONFIG_PORT_JSON_SERVER 19444                       // Port JSON
#define CONFIG_PORT_UDP_LED 19446                           // Port PROTO
 
#define CONFIG_WIFI_SSID "ssid"                             // Nom du Reseau Wifi
#define CONFIG_WIFI_PASSWORD "password"                     // Mot de Passe Wifi
#define OTA_PASSWORD "AmbiMod"                               // Mot de Passe pour mise a jour OTA

#define CONFIG_WIFI_STATIC_IP 1                             // Activation de l'addressage IP Statique. Commenter cette ligne pour passer en DHCP ou decommenter pour un adressage statique
  #ifdef CONFIG_WIFI_STATIC_IP                              // A modifier si activation de l'adressage IP Statique
    const byte CONFIG_WIFI_IP[] = {192, 168, 1, 123};
    const byte CONFIG_WIFI_SUBNET[] = {255, 255, 255, 0};
    const byte CONFIG_WIFI_DNS[] = {192, 168, 1, 1};
  #endif  
                                                                                                           // --- Configuration des LEDS --- //
#define PWM_LED                                            // A modifier en fonction de ce que vous avez avec la variable PWM_LED ou SPI_LED.
#define CONFIG_LED_SPI_CHIPSET APA102                      // Si vous utilisez une bande de LEDs SPI, remplacer par le modele que vous avez (APA102, WS2801...)
#define CONFIG_LED_CLOCKLESS_CHIPSET WS2812B               // Si vous utilisez une bande de LEDs PWM, remplacer par le modele que vous avez ( WS2812, WS2812B, WS2813...)

#define CONFIG_LED_COUNT 24                                // Nombre de Leds
#define CONFIG_LED_COLOR_ORDER RGB                         // Ordre des Couleurs
#define CONFIG_LED_DATAPIN D7                              // Broche pour le DATA
#define CONFIG_LED_CLOCKPIN D4                             // Broche pour la Clock si bande SPI.

// Si PWM_LED. Permet de regler quelques soucis.
//#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0 
                                                                                                            // ---  Autres Options --- //
                                                                                                            
#ifdef WEBCONFIG
  #define CONFIG_ENABLE_WEBCONFIG 1                        // Activation de la page de configuration WEB
  #define CONFIG_OVERWRITE_WEBCONFIG 1                     // Remplacement de toutes les valeurs de l'interface WEB par les valeurs defini ci-dessus. Commenter la ligne pour ne pas ecraser ces valeurs.
#endif

#define FASTLED_ESP8266_RAW_PIN_ORDER
//#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
//#define FASTLED_ESP8266_D1_PIN_ORDER
 
//OFF, HYPERION_UDP, STATIC_COLOR, RAINBOW, FIRE2012
#define CONFIG_LED_STANDARD_MODE RAINBOW     // Effet de Démarrage ou d'attente de liaison UDP 
#define CONFIG_LED_HYPERION_AUTOSWITCH true 
#define CONFIG_LED_STANDARD_MODE_TIMEOUT_MS 5000`

Sorry for the disturbance and thank you again for your wonderful works.

SciLor, 2018-05-11:
I think you confuse PWM and clockless. You seem to talk about clockless
You have got LED-Stripes with multiple independent selectable leds:
1. with clock/data (4-lines)
2. with just data (clockless, 3-lines)

You may have LED-Stripes, that are controlled directly via PWM (all LEDs have the same color)
3. 4-lines - GND, RED, BLUE, GREEN.

I would just place a
`#ifdef`
around the ConfigStatic.h as you did with webconfig.

Yous13, 2018-05-13:
Hi Scilor,

I change my mind after reading your last comment and i changed my Config static to use the 3 Types of LEDstrips you defined in you Configuration.

Here my Final Config Static with some reorganisation to make the editing simple as possible :

```
                                                                                                         // --- Web interface and Log Level --- //
#define WEBCONFIG                                           // Enable/Disable Web Interface ( Port 80 ). Comment this line to disable this function
#define LOGLEVEL LOG_LEVEL_INFOS                            // Enable/Disable LOG. Remplace LOG_LEVEL_INFOS by LOG_LEVEL_DEBUG or LOG_LEVEL_VERBOSE and Comment this line to disable LOG.

                                                                                    // ---  WIFI and OTA Configuration ( Can be changed on Web Interface ) --- //
#define CONFIG_WIFI_HOSTNAME "ESP8266"                      // Device name.
#define CONFIG_PORT_JSON_SERVER 19444                       // Port JSON.
#define CONFIG_PORT_UDP_LED 19446                           // Port PROTO.
 
#define CONFIG_WIFI_SSID "ssid"                             // Wifi Name.
#define CONFIG_WIFI_PASSWORD "password"                     // Wifi Password.
#define OTA_PASSWORD "AmbiMod"                              // Password for OTA programmation.

#define CONFIG_WIFI_STATIC_IP 1                             // Enable/Disable IP Static configuration . Comment this line to disable this function and activate DHCP. ( See below to set Static IP Address ).

// Hyperion start Effect.
#define CONFIG_LED_STANDARD_MODE RAINBOW                    // Change RAINBOW by OFF, HYPERION_UDP, RAINBOW, STATIC_COLOR or FIRE2012.
#define CONFIG_LED_HYPERION_AUTOSWITCH true 
#define CONFIG_LED_STANDARD_MODE_TIMEOUT_MS 5000

                                                                                       // --- LEDs Configuration. ( Cannot be changed on Web Interface  )  --- //
#define CONFIG_LED_CLOCKLESS                                // Set the type of LEDstipe you use between CONFIG_LED_CLOCKLESS, CONFIG_LED_PWM ( Non Adressable LEDstrips ) or CONFIG_LED_SPI.
#define CONFIG_LED_SPI_CHIPSET APA102                       // Change this value if you set CONFIG_LED_SPI ( APA102, WS2801... ).
#define CONFIG_LED_CLOCKLESS_CHIPSET WS2812B                // Change this value if you set CONFIG_LED_CLOCKLESS ( WS2812, WS2812B, WS2813... ).

#define CONFIG_LED_COUNT 24                                 // Number of LEDs. If CONFIG_LED_PWM, set the value to 1.
#define CONFIG_LED_COLOR_ORDER RGB                          // Color Order.
#define CONFIG_LED_DATAPIN D7                               // DATA Pin.
#define CONFIG_LED_CLOCKPIN D4                              // CLOCK Pin if CONFIG_LED_SPI_CHIPSET.

// Only if you set CONFIG_LED_CLOCKLESS/CONFIG_LED_PWM. Can fix Flickering, Stuterring...problems. Comment/Uncomment lines and/or change value to 1,2,3...
//#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0

#define FASTLED_ESP8266_RAW_PIN_ORDER                       // Can be changed with the value FASTLED_ESP8266_RAW_PIN_ORDER, FASTLED_ESP8266_NODEMCU_PIN_ORDER or FASTLED_ESP8266_D1_PIN_ORDER.

                                                                                      // ---  Optional options. Depending on your configuration set above --- //
// Change values if WEBCONFIG is active.
#ifdef WEBCONFIG                                            // NOT CHANGE THIS LINE.
  #define CONFIG_ENABLE_WEBCONFIG 1                         // NOT CHANGE THIS LINE.
  #define CONFIG_OVERWRITE_WEBCONFIG 1                      // Replaces all values in the webconfig with the configuration set in Config.Static
#endif

// Change values if Static IP Adress is activated.( Can be changed on Web Interface )
  #ifdef CONFIG_WIFI_STATIC_IP                              // NOT CHANGE THIS LINE.
    const byte CONFIG_WIFI_IP[] = {192, 168, 1, 123};       // IP Address.
    const byte CONFIG_WIFI_SUBNET[] = {255, 255, 255, 0};   // NetMask.
    const byte CONFIG_WIFI_DNS[] = {192, 168, 1, 1};        // DNS/Gateway.
  #endif  
  
// Change values if CONFIG_LED_PWM is set.
#ifdef CONFIG_LED_PWM                                       // NOT CHANGE THIS LINE.
  #define CONFIG_LED_PWM_RED D1                             // RED Pin.
  #define CONFIG_LED_PWM_GREEN D2                           // GREEN Pin.
  #define CONFIG_LED_PWM_BLUE D3                            // BLUE Pin.
#endif
```

And the part i changed on your WrapperLedControl.cpp
```
#ifdef CONFIG_LED_CLOCKLESS_CHIPSET
into
#ifdef CONFIG_LED_CLOCKLESS
```

I leave you now, i stop disturbing you on that. See you.

closed 2018-05-14 by SciLor