SciLor/Hyperion_LED-Controller #41: wifi AP

pete111, 2018-05-18:
Hello, 
please what happens when I leave blank wifi fields?
#define CONFIG_WIFI_SSID ""
#define CONFIG_WIFI_PASSWORD ""

How can I then access webinterface? Will it behave as standalone wifi AP then?
Because I did not find access point code in sources. Many thanks and sorry if my question is wrong.

SciLor, 2018-05-18:
Then nothing happens. There is no standalone WiFi AP function implemented.

pete111, 2018-05-18:
it would be nice to have standalone AP if no ssid and pass is saved in SPIFFS. And maybe to have an option to change AP ssid and pass also, so just simple storing of these values in SPIFFS as well :) It would be coool.

SciLor, 2021-05-08:
Implemented via https://github.com/SciLor/Hyperion_LED-Controller/commit/949a57b61cf51f69bfc742efbb7937734de7f58f

closed 2021-05-21 by SciLor