SciLor/Hyperion_LED-Controller #42: Run Hyperion only in one instance

DaPeace, 2018-06-04:
Is it possible to use hyperion only in one instance and forward the data to the esp8266 or is the second instance a MUST?

SciLor, 2018-06-04:
This is a hyperion related question, but you can easily use it with a single instance as primary output device
Just configurate hyperion as you would do it for a single instance and adjust the device configuration as described in the tutorials.

`{
      "colorOrder" : "rgb",
      "maxpacket" : 1450,
      "name" : "AmbiSZ-ESP8266",
      "output" : "ESP8266:19446", ///
      "protocol" : 0,
      "rate" : 250000,
      "type" : "udp"
},`

DaPeace, 2018-06-04:
Thanks for the quick answer. Will try that in the afternoon!

SciLor, 2018-06-06:
I am using it for 300 leds without any problem. You should check if the wifi connection ist strong enough. 
The color problem should be a problem with a wrong configuration. Please check the hyperion forum for that. Maybe the color order ist wrong.

DaPeace, 2018-06-06:
Yesterday i tried again without any luck. A small strip with 6 leds works but the 122 dont. Could it be that its not possible to drive the ws2812b-strip with 122 LEDS with the 3.3v-pins from the nodemcu?

SciLor, 2018-06-07:
You can drive as much leds you want to. You just need a fitting power delivery.
Please try the rainbow effects without hyperion itself. Sometimes there are problems driving the leds without a 5V converter for the led pins.

SciLor, 2018-06-08:
Better description see @https://github.com/SciLor/Hyperion_LED-Controller/commit/70adf21fd85af96dcd7423bb2cf44de30cd0cdfb

andrein81, 2020-04-05:
sorry for the silly question. How do you guys run 2 instances of hyperion?

closed 2021-05-21 by SciLor