SciLor/Hyperion_LED-Controller #43: Possibility to use protocol 2 on hyperion

DaPeace, 2018-06-05:
As stated here [(https://hyperion-project.org/wiki/UDP-Device)] protocol 2 would have some advantage to the raw format. At the moment in serial interface i see a complaint about wrong hashsize. Would it be possible to make that configurable between proto 1 and 2?

SciLor, 2018-06-06:
I would need to implement that protocol, but should be possible.

SciLor, 2018-06-10:
Protocol 2 is implemented as "UDP_FRAGMENT". But I merely see an advantage if the LED count is lower than 500 (as it will just get usefull if the packet size is bigger than 1500 bytes). I have also started to implement protocol 3 (TPM2), but it isn't ready yet!

Please check it if you can, it is active on the dev branch. Differences see https://github.com/SciLor/Hyperion_LED-Controller/compare/175c2e585ffdbf7a3631570cbc854dbf452b5ca9...99007eb83c48d7461d382e1f336b9fce61623801

Please also check if the webconfig is working.

menthon, 2019-01-28:
For a setup with ~600 pixels, and protocol 2 (UDP_FRAGMENT), only the first parts get updated (last part stays full white.

SciLor, 2019-01-28:
Could you enable debugging on verbose level and count the leds that are working?

menthon, 2019-01-28:
Ok, the following changes to the function 

> WrapperUdpLed::handleProtocolFragment(int bytes) 

in 

> WrapperUdpLed.cpp

 seem to have fixed that:

> For a setup with ~600 pixels, and protocol 2 (UDP_FRAGMENT), only the first parts get updated (last part stays full white.

for me.

`if (ledIdEnd <= _ledCount) {
      //_udp.read(_udpBuffer, bytes - 4);
      for (int i = ledIdStart; i < ledIdEnd; i++) {
        updateLed(i, _udp.read(), _udp.read(), _udp.read());  
      }
      refreshLeds();
      Log.verbose("Bytes available: %X . Flushing.", _udp.available());
      _udp.flush();
    } else {
      Log.debug("Too many LEDs: expected=%i, actual=%i", _ledCount, ledIdEnd);
    }`

SciLor, 2019-01-28:
I don't currently see what you have changed.
If you are unable to create a pull request please try to format the code, so the changes are clear. 
Thank you for your support and fix.

SciLor, 2019-01-29:
Thank you for your fix.
I currently don't understand why your code works, and mine not. I will investigate it. 
Your proposed fix is a bit of a hack, as you skip the buffers

SciLor, 2021-05-21:
https://github.com/SciLor/Hyperion_LED-Controller/commit/6090b06f6cf97d637229914d740df246a199b754

closed 2021-05-21 by SciLor