SciLor/Hyperion_LED-Controller #44: starting with solid color

maex-la, 2018-06-07:
i found the line #define CONFIG_LED_STANDARD_MODE in ConfigStatic.h

But where to define the static Color for this Standard mode?

Thank you!

SciLor, 2018-06-07:
According to https://github.com/SciLor/Hyperion_LED-Controller/search?utf8=✓&q=STATIC_COLOR&type=
I forgot to implement that. 
Thank you for your ticket.

maex-la, 2018-06-08:
Most usefull solution would be to start with the color last choosed (in Hyperion or anyway) before shut down. I don't know if this is possible. But would be great!

SciLor, 2018-06-08:
I think this would be a different topic. It would be quite difficult to implement that in a clean way, as I would need to save it to the flash and it doesn't like to be rewritten that often.
A workaround would be to just set it via the hyperion app, as you don't set it that often.

SciLor, 2018-06-08:
Implemented static color:
https://github.com/SciLor/Hyperion_LED-Controller/commit/3d5a974598ad77134a0f53645bbcdcc478a8d827

I am currently unable to test it myself, please test it.

maex-la, 2018-06-11:
Thank you.
I am not at home for several days. I will test it as soon as possible.

maex-la, 2018-06-14:
Tested it now.
The LED-Strip starts with White and remains so until a new color is picked in Hyperion App.

closed 2021-05-21 by SciLor