SciLor/Hyperion_LED-Controller #47: UPD Update Delay

osbash, 2018-10-13:
Hey!

So i'm having an issue with my setup.  I loaded up the controller to the ESP and configured the settings and the LEDs light up correctly.

I'm running a single instance of Hyperion and when I start it, the boot sequence displays on the LEDs a bit slow.  Then when I change the color, it takes about 30 seconds for the LEDs colors to change.  I also noticed that when I stop Hyperion, it takes about another 30 seconds for the LEDs to go back to the LED_STANDARD_MODE.

When checking the serial monitor in Arduino IDE, packets are still being received from Hyperion even after its stopped for about 30 seconds.  It seems like there is a huge delay but I can't figure out what is causing it.

**My setup is:**
ESP8266 NodeMCU LUA CP2102 ESP-12E
WS2812B
Testing on a strip of 38 LEDs
LED strip connected to Data Pin D1

**Hyperion Config**
`	"device" :
	{
		"name"       : "ESP-LEDs",
		"type"       : "udp",
		"output"     : "192.168.1.118:19446",
		"rate"     	 : 250000,
		"maxpacket"  : 1450,
		"protocol"   : 0,
		"colorOrder" : "rgb"
	},`

**Hyperion RGB Config**
#define CONFIG_LED_CLOCKLESS_CHIPSET WS2812B
#define CONFIG_LED_DATAPIN D1
#define FASTLED_ESP8266_RAW_PIN_ORDER

osbash, 2018-10-13:
**FIXED**

The root cause was the logging in ConfigStatic.h

When I commented out the logging, the speed was as expected!

SciLor, 2018-10-14:
Thank you, I will keep it open an make the README more clear about that. You could also lower the rate of hyperion, if there are delays left..

closed 2021-05-21 by SciLor