SciLor/Hyperion_LED-Controller #50: wifi connection problem

ageorgios, 2018-11-25:
I fail to connect to WiFi... ssid/password correct

```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
I:  Start logging
I:  CFG=Static config loaded
I:  Reset Mode
I:  Mode changed to 4
I:  Connecting to WiFi MG
I:  Using dynamic ip
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
E: WiFi Connection Failed! Rebooting...
```

any help?

SciLor, 2018-11-25:
Please set the logging to verbose.
Maybe a special char in you password

ageorgios, 2018-11-25:
```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
I:  Start logging
I:  CFG=Static config loaded
D:   Chipset=Clockless, dataPin=5, clockPin=NONE, colorOrder=66, ledCount=288
I:  Reset Mode
I:  Mode changed to 4
D:   Interval set to 16ms
D:   WrapperWiFi(ssid="MG", password="reallysimplepassword")
I:  Connecting to WiFi MG
I:  Using dynamic ip
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
E: WiFi Connection Failed! Rebooting...
```

not many new information with verbose though.

SciLor, 2018-11-25:
Please check if its a 2.4Ghz Wifi and if you can connect with a simple wifi example sketch with your esp8266/esp32

ageorgios, 2018-11-25:
It seems it is not associated with your implemenation
https://github.com/hrvach/espple/issues/4

closed 2018-11-25 by ageorgios