SciLor/Hyperion_LED-Controller #51: esp32 compilation steps

ageorgios, 2018-11-25:
what are the differences to compile to esp32?
I get this error

```
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\ageor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'D1' was not declared in this scope
```

ageorgios, 2018-11-26:
I succeeded compiling with 
```
#define CONFIG_LED_DATAPIN 5
```
although it core dumps
```
I:  Start logging
18:21:38.147 -> I:  CFG=Static config loaded
18:21:38.147 -> D:   Chipset=Clockless, dataPin=5, clockPin=NONE, colorOrder=66, ledCount=288
18:21:38.147 -> I:  Reset Mode
18:21:38.147 -> I:  Mode changed to 4
18:21:38.147 -> D:   Interval set to 16ms
18:21:38.147 -> D:   WrapperWiFi(ssid="MG", password="correct")
18:21:38.147 -> I:  Connecting to WiFi MG
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
18:21:38.444 -> Core 1 register dump:
18:21:38.444 -> PC      : 0x4008e82e  PS      : 0x00060934  A0      : 0x8008d9b2  A1      : 0x3ffb1a10  
18:21:38.444 -> A2      : 0x3ffcd1b4  A3      : 0x3ffb2050  A4      : 0x00000001  A5      : 0x00000001  
```

I have checked that with WiFi example it connects ok

SciLor, 2018-11-26:
I have tested it on a wemos esp32 dev module.
I think you have missed the right libraries first.

closed 2018-11-26 by SciLor