SciLor/Hyperion_LED-Controller #52: Question - Change digital pin state when leds are not working.

kiryl85, 2019-02-04:
Hi,

Would it be possible to add a new feature that would put specified digital pin to high when leds are not working for a specified length of time? Then it would be possible to control a relay that would turn off and on a power supply when it's not needed.

Cheers

SciLor, 2019-02-04:
I would suggest you to put a PIN to high just here (and in all other cases to low):

https://github.com/SciLor/Hyperion_LED-Controller/blob/589579726920aa14f096302d0e280ecbacd0fdf7/HyperionRGB/HyperionRGB.ino#L72

Use [digitalWrite](https://www.arduino.cc/reference/en/language/functions/digital-io/digitalwrite/) to set it.
You may need to initialize your PIN with [pinMode](https://www.arduino.cc/reference/en/language/functions/digital-io/pinmode/) just here
https://github.com/SciLor/Hyperion_LED-Controller/blob/589579726920aa14f096302d0e280ecbacd0fdf7/HyperionRGB/HyperionRGB.ino#L192

SciLor, 2019-02-04:
If you are using the off mode, it has to be set so. Normally you would use it, when the hyperion instance isn't connected, after x-milliseconds to shut of the leds.

kiryl85, 2019-02-04:
ok, but this works only when I stop hyperion
if you check `systemctl status hyperion` you will see that the message there changes whenever you start to play a video, start any effect or stop everything 

so my question is - is it possible to have some mecanism that would change pin mode after defined time while still being connected to hyperion instance? I'm sorry if I didn't make myself clear from the start.

I really apreciate your work :) Thank you

SciLor, 2019-02-06:
You will have to implement it yourself.

closed 2019-02-06 by SciLor