SciLor/Hyperion_LED-Controller #56: flickering?

deex2, 2019-05-09:
Hi,

I am currently building a setup with an ESP32 + APA102 + RPI0 w + Pi Camera.
The problem is that the leds are not stable at all. I am using the last version of Hyperion.ng that I have compiled on the RPI0.
At the beginning, I was thinking that it was a hardware problem, so I did multiple tests. At the end, it looks like a software problem.
Here some tests I did: if I turn the RPI0 + Hyperion, the ESP32 is showing a default color:
Here a video:
https://drive.google.com/file/d/1whaHCECnqdubt3Pj-SG7R2H-W-0Eexab/view?usp=sharing

As we can see here, leds are stable. Some this is not a hardware problem, or a esp32 problem.

If I turn on Hyperion.ng, the ESP32 is connected, and I ask Hyperion.ng to show an effect (flag for example). Here a video:
https://drive.google.com/file/d/1rcyTIvgrRD7z7GbB43no97DIXo5S4Ior/view?usp=sharing

As we can see, this is not stable at all, this is flickering. Any idea of why?

Thank you.

SciLor, 2019-05-09:
I havn't used an esp32 yet (so it is basicly untested). I am still using Hyperion not Hyperion.ng.

You may try to lower the baudrade and check if its fine. Maybe the wifi connection is a bit bad. You may also try to lower the loglevel or disable it.
If you have an ESP8266, please try with it. If you can install Hyperion instead of Hyperion.ng it would be also an option.

deex2, 2019-05-17:
So after hours of work, I tried Hyperion and not Hyperion.ng and..... still the same.
I tried to lower the rate to 50000 for example, no difference.
On the ESP32, I did:
`#define LOGLEVEL LOG_LEVEL_INFOS
//#define LOGLEVEL LOG_LEVEL_DEBUG
//#define LOGLEVEL LOG_LEVEL_VERBOSE`
Same result.

Any other idea that I can try please? (beside a ESP8266)
I would be happy to do more tests. I really want to find what is wrong.

SciLor, 2019-05-17:
I am sorry that I have no Idea why this is happening. Does it also happen when you are using the internal rainbow effect in my script (without hyperion)?
If yes, you may change the pinout to your leds. 
You may also try an esp8266, which should work fine.

deex2, 2019-05-17:
When I turn off the Rpi, and use only the esp32 with your code that shows a default rainbows from here: https://github.com/SciLor/Hyperion_LED-Controller/blob/master/HyperionRGB/ConfigStatic.h.example#L40
everything is stable, no flickering at all.

When I use Hyperion with the effect engine, this is flickering (so this is not the camera).

What is the difference between the internal rainbow and the rainbow in the effect engine of Hyperion? The one in Hyperion is updated by wifi continuously?

SciLor, 2019-05-17:
It has to be a problem with the WiFi connection.
Hyperion sends the led colors frame by frame.

deex2, 2019-05-20:
I will try with an esp8266 to have the same setup as you. Will be easier.
I will come back here and tell you.

deex2, 2019-05-23:
So, I tried with the same setup as you: esp8266 + regular Hyperion....and same flickering.
So I guess the problem doesn't come from the esp8266/esp32 or hyperion.ng/hyperion.
I did multiple tests to understand what is wrong.
Here the wiring:
https://image.jimcdn.com/app/cms/image/transf/dimension=1920x400:format=png/path/s01854e9c3e083509/image/i36cf2b90feaa7cd7/version/1485276753/image.png

The only difference with this image is that I connect the red line (+5v) to the VIN of the esp8266/esp32 (the board I use can take 5v as input: https://images-na.ssl-images-amazon.com/images/I/716lXq0diJL._SL1200_.jpg)

So here what I found: if I disconnect the +5v from the leds and connect an usb cable from my computer to the esp, I don't have any flickering (the leds are less bright but this is too test)!!!
So I don't understand. Any ideas please?

SciLor, 2019-05-23:
This is a hardware problem.
The problems seems to be the power supply or the signal levels to the leds. You may try to use a level shifter from 3.3V to 5V to connect the leds signal lines. 

For more help you should check out the hyperion forum or any other electronic forum

closed 2019-05-23 by SciLor

deex2, 2019-06-27:
So after doing more and more tests, using a custom code:
`#include <FastLED.h>

// How many leds in your strip?
#define NUM_LEDS 192

// For led chips like Neopixels, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806 define both DATA_PIN and CLOCK_PIN
#define FASTLED_ESP8266_RAW_PIN_ORDER
#define DATA_PIN 7
#define CLOCK_PIN 5

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
      FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, BGR>(leds, NUM_LEDS);
}

void loop() { 
  for( int j = 0; j < NUM_LEDS; j++) {
    // Turn the LED on, then pause
    leds[j] = CRGB::White;
    FastLED.show();
  }
  delay(10000);
  for( int j = 0; j < NUM_LEDS; j++) {
    // Now turn the LED off, then pause
    leds[j] = CRGB::Red;
    FastLED.show();
  }
  delay(10000);
}`

Here this is not flickering.
I created a message on reddit here: https://www.reddit.com/r/AskElectronics/comments/bystvc/leds_flickering_what_is_wrong_with_the_wiring/
where I give a lot of details.
Do you have any idea please?
I tried a signal level, resistors, etc etc...nothing really work.