SciLor/Hyperion_LED-Controller #57: Can hyperion extender be used with alexa..??

daksh1305, 2019-07-15:
Hello @SciLor,

I need some assistance with the project i was thinking to use hyperion extender with alexa so that it can control each standalone device with alexa.I researched about esp with alexa integration and found an project espalexa developed by @aircookie can espalexa be clubbed with the hyperion extender code.

Any help would be appreciated..:)

Thanks,

SciLor, 2019-07-15:
Hey Daksh,
I don't have an alexa device. So there is no interest and ressources on implementing https://github.com/Aircoookie/Espalexa for my project.

I am sorry that I can't help you with your project.

SciLor

closed 2019-07-15 by SciLor

pattyland, 2019-10-30:
FYI https://www.home-assistant.io/ has integrations for Alexa and Hyperion