SciLor/Hyperion_LED-Controller #67: Compile error in BaseHeader.h

Malvik298, 2021-08-30:
```
C:\Users\user\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  110 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
expected unqualified-id before '(' token 
```

In BaseHeader.h

SciLor, 2021-08-30:
Do you have some more details? You have just posted the Warning.

What lib versions you use? What error you get?

Malvik298, 2021-08-30:
I used the specified Libraries version and BoardVersion 3.0.1 (ESP8266) in ArduinoIDE 1.8.15.

Here's a full error log

```
Arduino: 1.8.15 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Serial, None, Only Sketch, 115200"

In file included from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\ConfigStructures.h:7,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\Config.h:9,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\BaseHeader.h:6,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.h:4,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.cpp:1:

C:\Users\malvik\Documents\Arduino\libraries\FastLED/FastLED.h:17:21: note: '#pragma message: FastLED version 3.001.007'

   17 | #    pragma message "FastLED version 3.001.007"

      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\malvik\Documents\Arduino\libraries\FastLED/led_sysdefs.h:27,

                 from C:\Users\malvik\Documents\Arduino\libraries\FastLED/FastLED.h:44,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\ConfigStructures.h:7,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\Config.h:9,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\BaseHeader.h:6,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.h:4,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.cpp:1:

C:\Users\malvik\Documents\Arduino\libraries\FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: conflicting declaration 'typedef uint8_t boolean'

   15 | typedef uint8_t boolean;

      |                 ^~~~~~~

In file included from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\BaseHeader.h:4,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.h:4,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.cpp:1:

C:\Users\malvik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:159:14: note: previous declaration as 'typedef bool boolean'

  159 | typedef bool boolean;

      |              ^~~~~~~

In file included from C:\Users\malvik\Documents\Arduino\libraries\FastLED/FastLED.h:68,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\ConfigStructures.h:7,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\Config.h:9,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\BaseHeader.h:6,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.h:4,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\LoggerInit.cpp:1:

C:\Users\malvik\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'

  110 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
[... 655 lines cut ...]
                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\ConfigStructures.h:7,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\Config.h:9,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\BaseHeader.h:6,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\WrapperWiFi.h:4,

                 from C:\Users\malvik\Desktop\Hyperion_LED-Controller-master\HyperionRGB\WrapperWiFi.cpp:1:

C:\Users\malvik\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'

  110 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

expected unqualified-id before '(' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

SciLor, 2021-08-31:
Please try the lib versions recommended in the Readme

Malvik298, 2021-09-07:
Well, it didn't worked. Couldn't spend much time debugging so closing this issue. Moved to WLED.

closed 2021-09-07 by Malvik298